GuLinux/AstroPhoto-Plus #17: Docker build fails

tkottary, 2018-07-29:
Could you suggest why is this failing . Attached complete logs.

  
       Name                        Command               State           Ports                                              
----------------------------------------------------------------------------------------                                      
starquew-dev_backend_1    /bin/sh -c /entrypoint           Exit 3                                                             
starquew-dev_frontend_1   /bin/sh -c /entrypoint           Exit 1                                                              
starquew-dev_nginx_1      nginx -g daemon off;             Up       0.0.0.0:3000->80/tcp                                        
starquew-dev_redis_1      docker-entrypoint.sh redis ...   Up       6379/tcp    
~~~
backend_1   |   File "/app/star-quew.py", line 2, in <module>
backend_1   |     from api_decorators import *
backend_1   |   File "/app/api_decorators.py", line 2, in <module>
backend_1   |     from api_utils import *
backend_1   |   File "/app/api_utils.py", line 2, in <module>
backend_1   |     from controller import controller
backend_1   |   File "/app/controller.py", line 3, in <module>
backend_1   |     from models import Server, Device, Property, SavedList, INDIService, Sequence, INDIProfile, settings
backend_1   |   File "/app/models/__init__.py", line 2, in <module>
backend_1   |     from .server import Server
backend_1   |   File "/app/models/server.py", line 1, in <module>
backend_1   |     from pyindi_sequence import INDIClient
backend_1   | ModuleNotFoundError: No module named 'pyindi_sequence'
backend_1   | [2018-07-29 15:13:01 +0000] [114] [INFO] Worker exiting (pid: 114)
backend_1   | [2018-07-29 15:13:01 +0000] [111] [INFO] Shutting down: Master
backend_1   | [2018-07-29 15:13:01 +0000] [111] [INFO] Reason: Worker failed to boot.
starquew-dev_backend_1 exited with code 3
~~
[docker-build-logs.txt](https://github.com/GuLinux/StarQuew/files/2238906/docker-build-logs.txt)

GuLinux, 2018-07-29:
Hi,
as stated in the readme, you should first initialize all the git submodules:
```
git submodule init
git submodule update
```

The docker-compose stuff is mainly for development with simulators though, are you trying to develop, or do you just want to start the app?
If it's the latter, I did setup a circleci job publishing [here](https://gulinux.net/downloads/StarQuew/) daily builds, and with setup instructions in the [wiki](https://github.com/GuLinux/StarQuew/wiki/Installation).

p.s.: I was just about to publish raspberry pi ready images, and a first "public" beta version :)

GuLinux, 2018-07-29:
Sorry, my bad.. I think it *was* in the readme, but I might have removed it when I moved most of the docs in the wiki

tkottary, 2018-07-29:
Works perfectly fine now ! Sorry i thought i misread the read me too :)  
I want to contribute to this project at some point or may be help testing few things. I was able to install the app directly on the tinkerboard running  couple of days ago but it needed another library to be installed.  I did not want to open a issue for it looking at the commit rate.

Excited to try the beta version :)  Again big thank you for this project.

closed 2018-07-29 by tkottary

GuLinux, 2018-07-29:
Good to know, I really hope this project might be interesting to people. And I think the fact that it will run very smoothly on a raspberry pi (headless, no X11 setup) might be a very attractive feature.

I already have a very long list of things to do, so whenever you feel comfortable contributing just ping me :) The first major things I'd like to implement are FWHM (or even better HFD) for focus helping, and an autoguider.