GuLinux/AstroPhoto-Plus #18: When I try to run the git codes, frontend show that server error

umi-neko, 2018-07-31:
Frontend build with no error. Redis and backend already running, then start frontend, the browser show following message:

Server error

It looks like your backend server is not running, or encountered a fatal error.

Please check your backend server is up and running, and try reloading this page.
Error details:

Error source: network_request

exception: SyntaxError JSON.parse: unexpected character at line 1 column 1 of the JSON data

OS: archlinux
I am running backend in debug mode, but cann't see any error message.

GuLinux, 2018-07-31:
Hi,
I should probably make an AUR version, since I have ArchLinux as well :D
Did you also install a web server (nginx/apache2) with the relative configuration? It looks like the backend is not accessible, so that might be the reason

umi-neko, 2018-07-31:
Thanks for this information, I will try nginx later

GuLinux, 2018-07-31:
I added a sample `nginx` configuration for development in here: https://github.com/GuLinux/StarQuew/tree/master/config/dev-webserver-conf/nginx
Please check it out and let me know :)

umi-neko, 2018-08-03:
Hi GuLinux, I used the nginx configuration you added, but the frontend still show the same error page.

I have try GET command from the browser address bar, such as "http://localhost:5000/api/server/status" and "http://localhost:5000/api/version", it seems that backend can return correct JSON data.

then I open debug console in Firefox, and find an error message "Firefox can't make a connection to http://localhost:3000/api/events    events.js:82". The port number 3000 confused me, doesn't that should be 5000?

So what can I do now? I am not familiar with javascript. Or maybe I should give raspberrypi a try. :(

GuLinux, 2018-08-03:
Hi,
I am guessing you're trying just to use the application, not to develop on
it.  If that's the case, I'll make an aur package as soon as I can (today
probably). The configuration you're trying to run is mainly meant for
development, I guess this might be why you're having troubles.

I'm also going to release 0.1.1 today, with a few interesting improvement
(particularly for the raspberry pi)

umi-neko, 2018-08-03:
Thank you very much

GuLinux, 2018-08-03:
Hi,
quick test if you can:
if you grab this build: https://gulinux.net/downloads/StarQuew/ci-206/StarQuew-0.1.1-b206-Linux.tar.gz

Unpack it, go to the `StarQuew-0.1.1-b206-Linux/lib/StarQuew/backend/` directory, and then run `./start-server`, then open your browser to `http://localhost:5000`, everything should automatically work.

I'll still need to make a package of course (for dependencies, automatic systemd startup, etc)

GuLinux, 2018-08-03:
Just release v0.1.1 with a bunch of usability fixes, and also the ability of running standalone, without nginx.
You can also install from AUR: https://aur.archlinux.org/packages/starquew/

Just install the package, follow the instructions on the console (I'll update the wiki with the *Other Distributions* section) and point your browser to http://localhost:5000

umi-neko, 2018-08-08:
Hi GuLinux, I install it from AUR, and it worked now. thank you very much. :+1:

closed 2018-08-08 by umi-neko