GuLinux/AstroPhoto-Plus #116: Support for RPi0W

universe241981, 2019-02-24:
Indilib 1.7 and above not supported in RPi0W. I have manually built the indilib 1.6.2 (highest version that work with RPi0W) in the AstroPhoto-Plus Raspbian img and it work on RPi0W.
Then another issue came. The image appear to be not debayered correctly with image overlaping.
Anyway, could you add support for RPi0W？ Thanks.
[image]

GuLinux, 2019-02-24:
Hi,
*AstroPhoto Plus* should in theory support the Raspberry Pi Zero W (at its core it's just a python backend plus a web javascript application).

Basically if you install all the INDI dependencies, installing *AstroPhoto Plus* should only require you to install the `.deb` file on your system, and configure it with `AstroPhotoPlus autosetup`.

However it is a very low powered device, with a single usb port, and thus is not exactly the ideal target for the application, so using it is not recommended. If by "support" you mean creating an image ad hoc for the Zero W, then no, it's not gonna happen, as the device is way too underpowered to be added as a supported device, and also we have a hard dependency on libindi: if they stopped supporting the Zero W from version 1.7, they probably have good reasons, and I can't support a device that is not support by the libraries I depend on.

As for the debayering issue, please open another ticket.
Please specify the camera you're using, and attach a FITS file as it comes from your camera, and possibly logs as well (set logging level to `DEBUG` in Settings, then run `AstroPhotoPlus-ctl logs` on a shell on the server).

I would also really recommend trying *AstroPhoto Plus* on another device, just in case there's an issue with some Zero W library.

closed 2019-02-24 by GuLinux

universe241981, 2019-02-24:
Hi, thank you for your replies.
Initially, I am afraid on conflict to have two different version indilib in the Raspbian img you created.
But since it seem got no problem at the moment to use Pi0W for image capture, so this Pi0W support issue can be ignored.
I will take some images for you to review it. Thanks.