GuLinux/AstroPhoto-Plus #117: Easy setup guide for RPi (Raspbian) package: command error?

universe241981, 2019-02-24:
I have Raspbian stretch lite on RPi3 but failed to do easy setup by referring to the website.
wget -N -q https://raw.githubusercontent.com/GuLinux/AstroPhoto-Plus/master/scripts/setup/raspbian.sh && sudo bash ./ubuntu.sh && rm raspbian.sh
Anything wrong with the command? Or that is my RPi3 issue?

GuLinux, 2019-02-24:
Hi, would you please attach some logs of the installation script?

universe241981, 2019-02-24:
I am not sure about the logs of the installation script. But I have screenshot it here. I am running the installation on RPi3B.
[image]
[image]
[image]

GuLinux, 2019-02-24:
To redirect logs to a file, just run a command like this:
```
COMMAND 2>&1 | tee command.log
```
(of course replace `COMMAND` with the command you want to run).
Try also running one of the following:
```
systemctl status nginx
journalctl -xu nginx
```
And attach the logfile in `/var/log/nginx/error.log`

universe241981, 2019-02-25:
I found the error on the command that provided in the website for easy setup of RPi (Raspbian) package.

The original command in the website is:
`wget -N -q https://raw.githubusercontent.com/GuLinux/AstroPhoto-Plus/master/scripts/setup/raspbian.sh && sudo bash ./ubuntu.sh && rm raspbian.sh`
The "bash ./ubuntu.sh" is the error command.

It should be:
`wget -N -q https://raw.githubusercontent.com/GuLinux/AstroPhoto-Plus/master/scripts/setup/raspbian.sh && sudo bash ./raspbian.sh && rm raspbian.sh`
The "bash ./ubuntu.sh" need to change to "bash ./raspbian.sh".

Thank you very much.

GuLinux, 2019-02-25:
Thanks for reporting, just fixed.
Was this the cause of your errors? Can the issue be closed?

universe241981, 2019-02-26:
The previous command error cause the command cannot be executed. After corrected, it can run as normal and install everything needed fro the AstroPhotoPlus.

Anyway, about the issue of not able to configure nginx, the reason is the ipv6. As my Raspbian that run on RPi3 has ipv6 disabled, thus it cause nginx cannot be configure. Once I set the ipv6 back on for RPi3, then it is ok.

So now the remaining issue is the wait time of the "wait for backend server to be ready..."
My suggestion is to put a statement in your website or anywhere else in order for RPi user alert about this.

Thank you.

Yes. this issue can be closed.

GuLinux, 2019-02-26:
Thanks,
Yes, I'll add it to my FAQs page I guess.
Cheers,
Marco

closed 2019-02-26 by GuLinux