GuLinux/AstroPhoto-Plus #122: Color camera debayer problem

universe241981, 2019-02-24:
Camera: QHY5P-II-C
Sample FITS image attached

The FITS image look good but when display as jpg in the web interface, it not good.

[log.txt](https://github.com/GuLinux/AstroPhoto-Plus/files/2897912/log.txt)
[fits-header.txt](https://github.com/GuLinux/AstroPhoto-Plus/files/2897913/fits-header.txt)

GuLinux, 2019-02-25:
Thanks, I found the root cause in a bad data management in 8bit mode (which I very rarely use).
I need a bit of time to fully verify the fix works in both 8bit and 16bit mode, in the meantime I would recommend using 16bit only until this gets fixed.
Cheers,
Marco

universe241981, 2019-02-26:
16 bit mean select png format instead of jpg format?

GuLinux, 2019-02-26:
Hi,
No, it's a camera control, I believe it should be found under the `Controls` group of the INDI control panel for the camera

GuLinux, 2019-02-27:
Sorry, haven't had time to check on my camera :(
There should be a "Format" INDI switch, you can see an example here:
https://goo.gl/images/z47et5

closed 2019-02-27 by GuLinux

GuLinux, 2019-02-27:
Hi, please have a go at testing the new version, you can just run the auto updater, and verify that the build version is 640, or just download the package for said build from here: https://astrophotoplus.gulinux.net/development-builds

universe241981, 2019-02-28:
Hi,
I have update the build to version 640 by download the package.
Initially, I tried to update via auto updater in the web interface but failed. Then only use the downloaded package for update.
Everything work fine now.
Thank you.

commit 3a21d82f (2019-02-27): Correctly handle both 8 and 16 bit images; better debayering mapping
Fixes #122