GuLinux/AstroPhoto-Plus #123: Waiting for backend server to be ready...

universe241981, 2019-02-25:
Usually how long it need to be wait? 
What cause the long wait?
As of the moment i writing this, it still "Waiting for backend server to be ready..."
Thanks.

GuLinux, 2019-02-25:
On a slow machine (like the raspberry pi for instance, or even worse the Pi Zero) it can take quite a lot, as it has to install all the python dependencies, and compile native modules.
I'm not sure how long, but it might even be more than half hour on a slow system.
You can check the backend status by running
```AstroPhotoPlus-ctl logs```

closed 2019-02-25 by GuLinux

universe241981, 2019-02-26:
It seem that not possible to setup in Pi0W...?
[Pi0W setup_not complete.txt](https://github.com/GuLinux/AstroPhoto-Plus/files/2903899/Pi0W.setup_not.complete.txt)