GuLinux/AstroPhoto-Plus #125: Easy setup for RPi (Raspbian) package installation issue

universe241981, 2019-02-27:
Sorry to re-open a ticket about the easy setup installation for RPi (Raspbian) package.
I am using RPi3b

After command `wget -N -q https://raw.githubusercontent.com/GuLinux/AstroPhoto-Plus/master/scripts/setup/raspbian.sh && sudo bash ./raspbian.sh && rm raspbian.sh`

The installation run automatically until setup of backend server then it failed.
Reboot and command `sudo AstroPhotoPlus-ctl autosetup`
backend server setup ready but no wifi access point setup available.

Reboot again and command `wget -N -q https://raw.githubusercontent.com/GuLinux/AstroPhoto-Plus/master/scripts/setup/raspbian.sh && sudo bash ./raspbian.sh && rm raspbian.sh`
Then only can setup the wifi access point.

Is that my RPi issue or the installation has issue?
I attach some logs for your reference.

[Installation history summary.txt](https://github.com/GuLinux/AstroPhoto-Plus/files/2908452/Installation.history.summary.txt)

GuLinux, 2019-02-27:
Hi,
The installation per se should be fine on a "vanilla" raspberry pi. In fact, I use exactly the same script to create the pre-customised image, and it doesn't seem to have issues.

Of course, being a simple bash script, it might not be perfect at detecting various edge cases, so there's surely lots of room for improvement.

I'm not sure why it is failing on your installation, I can find only 2 points where there's something weird happening, but I can't think of a cause for them.

The first is the redis unit not being enabled:
```
Failed to enable unit: Refusing to operate on linked unit file redis.service
```

The second is this stacktrace:
```
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]: Exception:
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]: Traceback (most recent call last):
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:   File "/home/pi/.local/share/AstroPhotoPlus/python-venv/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 594, in urlopen
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:     chunked=chunked)
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:   File "/home/pi/.local/share/AstroPhotoPlus/python-venv/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 391, in _make_request
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:     six.raise_from(e, None)
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:   File "<string>", line 2, in raise_from
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:   File "/home/pi/.local/share/AstroPhotoPlus/python-venv/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 387, in _make_request
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:     httplib_response = conn.getresponse()
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:   File "/usr/lib/python3.5/http/client.py", line 1198, in getresponse
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:     response.begin()
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:   File "/usr/lib/python3.5/http/client.py", line 297, in begin
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:     version, status, reason = self._read_status()
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:   File "/usr/lib/python3.5/http/client.py", line 266, in _read_status
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:     raise RemoteDisconnected("Remote end closed connection without"
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]: http.client.RemoteDisconnected: Remote end closed connection without response
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]: During handling of the above exception, another exception occurred:
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]: Traceback (most recent call last):
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:   File "/home/pi/.local/share/AstroPhotoPlus/python-venv/lib/python3.5/site-packages/pip/basecommand.py", line 215, in main
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:     status = self.run(options, args)
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:   File "/home/pi/.local/share/AstroPhotoPlus/python-venv/lib/python3.5/site-packages/pip/commands/install.py", line 353, in run
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:     wb.build(autobuilding=True)
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:   File "/home/pi/.local/share/AstroPhotoPlus/python-venv/lib/python3.5/site-packages/pip/wheel.py", line 749, in build
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:     self.requirement_set.prepare_files(self.finder)
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:   File "/home/pi/.local/share/AstroPhotoPlus/python-venv/lib/python3.5/site-packages/pip/req/req_set.py", line 380, in prepare_files
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:     ignore_dependencies=self.ignore_dependencies))
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:   File "/home/pi/.local/share/AstroPhotoPlus/python-venv/lib/python3.5/site-packages/pip/req/req_set.py", line 554, in _prepare_file
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:     require_hashes
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:   File "/home/pi/.local/share/AstroPhotoPlus/python-venv/lib/python3.5/site-packages/pip/req/req_install.py", line 278, in populate_link
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:     self.link = finder.find_requirement(self, upgrade)
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:   File "/home/pi/.local/share/AstroPhotoPlus/python-venv/lib/python3.5/site-packages/pip/index.py", line 465, in find_requirement
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:     all_candidates = self.find_all_candidates(req.name)
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:   File "/home/pi/.local/share/AstroPhotoPlus/python-venv/lib/python3.5/site-packages/pip/index.py", line 423, in find_all_candidates
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:     for page in self._get_pages(url_locations, project_name):
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:   File "/home/pi/.local/share/AstroPhotoPlus/python-venv/lib/python3.5/site-packages/pip/index.py", line 568, in _get_pages
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:     page = self._get_page(location)
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:   File "/home/pi/.local/share/AstroPhotoPlus/python-venv/lib/python3.5/site-packages/pip/index.py", line 683, in _get_page
[... cut ...]
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]:     total -= 1
Feb 27 09:58:40 raspberrypi AstroPhoto-Plus[13341]: TypeError: unsupported operand type(s) for -=: 'Retry' and 'int'
Feb 27 09:58:41 raspberrypi systemd[1]: AstroPhotoPlus.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
```

Although the stacktrace seems more worrying, the redis issue made me think of something: maybe you're not using raspbian stretch? If so it might be a distribution version compatibility issue.

Thanks,
Marco

universe241981, 2019-02-27:
I download the Raspbian Stretch lite from Raspberry Pi official website.
What I did is change the locale setting and apt upgrade the lsb_release from 9.4 to 9.8.
Then I execute your command for AstroPhoto Plus installation.
Perhaps I can redo all these step and copy all for you as reference.

Thank you very much.

universe241981, 2019-02-28:
Hi, I recorded all the process of installation from the Raspbian Stretch update until AAP installed.

Raspbian Stretch system update:
[log 1 - system update.txt](https://github.com/GuLinux/AstroPhoto-Plus/files/2914320/log.1.-.system.update.txt)

1st initial APP installation and AstroPhotoPlus-ctl logs. It stucked. More than 30 mins without further progress.
[log 1 - 1st initial AAP installation.txt](https://github.com/GuLinux/AstroPhoto-Plus/files/2914324/log.1.-.1st.initial.AAP.installation.txt)
[AstroPhotoPlus-ctl logs for 1st initial AAP installation.txt](https://github.com/GuLinux/AstroPhoto-Plus/files/2914325/AstroPhotoPlus-ctl.logs.for.1st.initial.AAP.installation.txt)

Power off the RPi3b and power up again. Then Repeat the APP installation.
[log 2 - reboot n re-install AAP.txt](https://github.com/GuLinux/AstroPhoto-Plus/files/2914329/log.2.-.reboot.n.re-install.AAP.txt)
[AstroPhotoPlus-ctl logs for reboot n re-install AAP.txt](https://github.com/GuLinux/AstroPhoto-Plus/files/2914330/AstroPhotoPlus-ctl.logs.for.reboot.n.re-install.AAP.txt)

This is how I install the APP.

universe241981, 2019-03-03:
Hi, I have retry to install AstroPhoto-Plus with following command:
`wget -N -q https://raw.githubusercontent.com/GuLinux/AstroPhoto-Plus/master/scripts/setup/raspbian.sh && sudo bash ./raspbian.sh && rm raspbian.sh`

This time, the installation going smooth. Attached some log for your reference:
[AstroPhotoPlus-ctl logs.txt](https://github.com/GuLinux/AstroPhoto-Plus/files/2922918/AstroPhotoPlus-ctl.logs.txt)
[Install logs.txt](https://github.com/GuLinux/AstroPhoto-Plus/files/2922919/Install.logs.txt)

GuLinux, 2019-03-04:
Thanks!
There's definitely room for improvements on the install script, but the fact that the installation process went smoothly kinda comforts me.
I'll probably do another intensive round of checks around the next release, so I'll keep this ticket open as a reminder.
Cheers,
Marco

universe241981, 2019-03-06:
As this issue has been improved in b420 release, thus it should be appropriate to close this ticket.
What do you think?

GuLinux, 2019-03-06:
As I mentioned, I prefer to keep it open as a reminder to do a full audit of the installation process for the next major release :)

GuLinux, 2020-03-30:
Hi, closing as this should now be properly fixed in recent build.

closed 2020-03-30 by GuLinux