GuLinux/AstroPhoto-Plus #133: Automation on repeat sequence capture function for All sky monitoring

universe241981, 2019-03-04:
All sky monitoring could be a 24/7 sequence image capturing job.
After setup the sequence capture job, the only thing to do is to repeat the same sequence daily.
Wonder how this could be possible?

Example of my all sky monitoring plan:
1st sequence job -- 5 secs exposure with 1 min interval between frames. 12 hrs nighttime capture.
2nd job -- Change gain & exposure time setting in prepare for daytime sequence capture.
3rd sequence job -- 1/500 secs exposure with 1 min interval between frames. 12 hrs daytime capture.
4th job -- Change gain & exposure time setting in prepare for nighttime sequence capture.
--> repeat the 1st sequence job

GuLinux, 2019-03-04:
Hi,
Well, in theory I might write a simple "Reset and Restart" sequence job.

There would be an issue with that though: sunrise/sunset calculation.
You obviously wouldn't want *exactly* 12 hours day and 12 night, but you want to be capable of adjusting the sequence duration based on sunrise and sunset.

I think your best shot is to use a script (bash, python, whatever you prefer) to call the Backend API directly.
You can setup two different sequences (day and night), and then start/stop/reset them with your script using the APIs you can find in this file: https://github.com/GuLinux/AstroPhoto-Plus/blob/master/backend/api_entrypoint.py
(unfortunately it's not documented, sorry about that).

For instance, to start a sequence, you can just run
```
curl -X POST http://<astrophoto-plus-ip>/api/sequences/<sequence-id>/start
```

I hope this helps

universe241981, 2019-03-06:
Hi,
The place where I live has just 30 minutes of changes between day & night duration. Thus, a 12 hrs interval for both day & night setting should work to me.
Thus, I think a simple "Reset and Restart" sequence job should work ok to me.

By the way, as All Sky images no need to be in RAW format. Most of the case should be just jpg format. I wonder if there could be an option to choose between RAW or jpg as output?
Another by the way, I found that if the sequence job can also save a jpg file of the images taken (not thumbnail preview jpg file), that would be nice too.

GuLinux, 2019-03-14:
Right, I'll see to add this into the backlog, it should be fairly simple.

GuLinux, 2019-03-14:
As for the format, I'll reply in the other issue

universe241981, 2019-03-22:
Hi, I've tried to use `curl -X POST http://<astrophoto-plus-ip>/api/sequences/<sequence-id>/start` to start a sequence job. But seem nothing happen after I command in terminal.
If not mistaken, if I have setup a sequence job called `test_sequence`, the command should be something like this `curl -X POST http://192.168.x.x/api/sequences/test_sequence/start`
Is that correct?

GuLinux, 2019-03-22:
Hi,
No, the sequence id is wrong, it doesn't correspond to the sequence name.
It's the alphanumeric string that you can find when you navigate to the sequence details page, in my case the url is `http://localhost:1080/sequences/ba547dadf0164c9983c0f765319ea43c`, hence the sequence id is `ba547dadf0164c9983c0f765319ea43c`.
You can also spy the API calls using your browser development tools (in the network tab).
I really should document the REST APIs as soon as possible :)

universe241981, 2019-03-26:
Hi, I've tried the command.
It only work for stop and start the sequence job but not reset.
Below is what I got when I tried to reset it:
`curl -X POST http://localhost:80/api/sequences/e21390f5b3b94170a9c32b120eb6dfba/reset
{"error":"Bad Request","error_message":"Invalid JSON request","payload":null}`

Any idea what happen? Thanks.

GuLinux, 2019-03-28:
That's because some APIs need parameters in form of a JSON object.
You can see that in the [api_entrypoint.py](https://github.com/GuLinux/AstroPhoto-Plus/blob/master/backend/api_entrypoint.py) file, they're marked with the `@json_input` annotation.
In this case you need to specify a `remove_files` parameter, `true` if you want to remove all the files saved by the sequence (I guess for you it's false).
It should look like this:
```
curl -X POST http://localhost:80/api/sequences/e21390f5b3b94170a9c32b120eb6dfba/reset -H 'Content-Type: application/json' -d '{ "remove_files": false }'
```

universe241981, 2019-03-30:
Please forgive me on knowing nothing about coding...
I've tried the command of Reset that you mentioned in previous latest reply. It works!

Just got a minor question here:
- First time Reset by command, the sequence still in `stopped` status. If I chose to set the `remove_files` parameter as `true`, files deleted. But the status of the sequence is `stopped`. Not `Idle`.
- Second time Reset by command, then only the sequence status set to `Idle`.

What do you think about this behavior? Normal?