GuLinux/AstroPhoto-Plus #134: How to Uninstall?

timbeccue, 2019-03-05:
I can't figure out how to uninstall AstroPhotoPlus. It would make sense to have that as an option as a command, eg `AstroPhotoPlus-ctl uninstall`. 

In the meantime, could you please direct me on what steps I need to take to uninstall? Thanks.

GuLinux, 2019-03-05:
Hi,
I guess it depends on how you did install it. I cannot put an uninstall command into the `AstroPhotoPlus-ctl` script, because the installation is dependent on the own distribution packaging system.

If you used the `deb` package, then it's done via `dpkg`.
You also should remove cache/config/shared files in your home (`$HOME/.local/share/AstroPhotoPlus`, `$HOME/.cache/AstroPhotoPlus`, `$HOME/.config/AstroPhotoPlus`) but that's an entirely optional step though (I might put this one in the `AstroPhotoPlus-ctl` script).

May I ask you why uninstalling? 
Cheers,
Marco

timbeccue, 2019-03-05:
Thanks for the direction; uninstalled smoothly.

I was curious about the program and wanted to try it out, but don't have an actual use for it at the moment. Wanted to keep my computer clean in the meantime. It looks like a great tool if I ever need it!

closed 2019-03-05 by timbeccue