GuLinux/AstroPhoto-Plus #135: Camera exposure time limited to certain number

universe241981, 2019-03-05:
Device: QHY5P-II-C

Exposure time more than 3 secs will cause TIME OUT of the exposure task. Exposure is in busy mode and  not complete.
Tried various setting such as USB speed & traffic. Still the same behavier.
Hopefully this is not the specific problem of QHYCCDs.

[AstroPhotoPlus-ctl logs.txt](https://github.com/GuLinux/AstroPhoto-Plus/files/2930106/AstroPhotoPlus-ctl.logs.txt)
[Error Message.txt](https://github.com/GuLinux/AstroPhoto-Plus/files/2930107/Error.Message.txt)

GuLinux, 2019-03-05:
Hi,
I think QHY cameras have a "Long exposure" setting that you have to enable to be able to take images longer than 1/2 seconds.
Could you please verify that?

Also, if you go to INDI control panel, could you check if eventually the control `Exposure` stops being busy at some point, or just gets stuck there?

Another test you can try is verifying with another INDI client (for instance Ekos) if the behaviour is the same or not.

I would say this might be a specific QHY problem though, as ASI cameras work pretty well in my case, and QHY unfortunately has quite a long tradition of driver problems.

Cheers,
Marco

universe241981, 2019-03-06:
Hi Marco,
After re-install the AstroPhoto-Plus, the QHY5P-II-C can do exposure up to 30 seconds. I didn't test further than 30 secs as my use case only up to 30 secs is enough.
Anyway, the initial test has TIMEOUT error a few times as I am changing the USB traffic setting and exposure time for try and error.
After a reboot and fix the USB traffic at 5, the other setting remain unchanged, I can do various exposure from 2 secs to 30 secs without problem.
All the above tested on RPi3b. No "long exposure" setting found. The "Exposure" section on the INDI control panel is stop (NOT "busy" stage) after the TIMEOUT happen.

By the way,
I've also experimented the same thing on Pi0W.
In order for Pi0W to work with AstroPhoto-Plus, the INDI need to use the version 1.5.0.
Download the INDI v1.5.0 and built accordingly. But I only built & install the general indilib, libqhy & indi-qhy.
Then the QHY5P-II-C can do the same thing as what I did on RPi3 that mentioned above.

closed 2019-03-06 by universe241981