GuLinux/AstroPhoto-Plus #136: Wifi client setup and usage

universe241981, 2019-03-07:
May I know what is the function of the wifi client setup?
Is that mean can connect to a local network to go into AstroPhoto-Plus web interface?

By the way, is there any possibility to use local network instead on static AP IP mode?

Thanks.

GuLinux, 2019-03-17:
Hi, sorry, I missed this issue before.
In a way it's more a configuration than an actual feature. It's available only a raspberry pi at the moment (mainly because it uses dhcpcd instead of network manager and similar).
It's just an interface to switch between AP mode (Wifi Access Point) and client mode (connecting to an existing access point).

What do you mean by "use local network instead on static AP IP mode" exactly?

universe241981, 2019-03-18:
What I mean is actually, if I am at home with existing access point (with internet access too), I would like the Pi connected to the home access point. While I am out from any internet access point, I would like the Pi act in AP mode so that I can access the Pi with laptop or mobilephone.

So you mean this feature of switch between AP mode (Wifi Access Point) and client mode (connecting to an existing access point) is not available on Pi?

GuLinux, 2019-03-18:
No, it's the other way around: this feature *is* available on the raspberry pi, but not on other clients.
Mainly because on a raspberry pi I can safely make some assumption (what kind of network manager is there), while of course on other platform things can vary quite a lot.

I need to write some user guide for it of course, but here's how you can configure it for your scenario:

You can toggle on/off the access point mode by clicking the "Turn On (or off) Access point" button, in the Settins/Commands section.
When you turn it on, you'll be broadcasting with the ESSID/PSK configured by the "Configure Access Point" button (the raspbian installation script lets you configure it, if you download the image it's `AstroPhoto-Plus`/`AstroPhoto-Plus`).
When you turn it off, it's gonna connect to the Access Point configured with the "Configure Wifi Client" button.

You just need to make sure you set it to the appropriate mode *before* you move the raspberry pi outdoors or indoors.

This will need some rework obviously, but it's a good starting point.
I'll need to find a more generic solution (perhaps based on network manager when available), and more importantly a semi automatic rollback procedure (if you can't connect to any access point, switch back to access point mode).

Anyway I wanted to ask you, if you wish to collaborate in any way, even just by writing docs, it's more than welcome. For this part for instance, you might write a tutorial for Raspberry Pi usage.
All docs on website are in the markdown format, so it's quite easy to write.

Cheers,
Marco

universe241981, 2019-03-19:
Hi Marco,
I would be more than happy to do something on this.
Let see how I can write some user experience on AstroPhoto-Plus on RPi.

GuLinux, 2019-03-19:
Cool, just let me know if you need anything :)
You have my email I guess, and you can also find the new webchat if you like in here: https://riot.im/app/#/room/#astrophotoplus:matrix.org
Cheers,
Marco

universe241981, 2019-05-16:
Dear Marco,
I have tried the function of network mode switching interface on RPi.
As what you described above, after the configuration of Wifi Access Point has done, turn off the RPi AP mode, then restart the RPi.
The RPi will search for Wifi Access Point according to ESSID/PSK that has been configured earlier.

If wanna switch back to RPi AP mode, just click the "Turn On (or off) Access point" button and restart the RPi.

I will close this ticket from now as it is not an issue anymore.
Thanks.

closed 2019-05-16 by universe241981

GuLinux, 2019-05-16:
Cheers :)
It still probably needs to be documented properly, and possibly simplified
a little bit, but good to know that it's working as expected :)