GuLinux/AstroPhoto-Plus #139: add change of port for browser

dragonlost, 2019-03-18:
Hello.
The software is accessible via port 80 in web browser.
I would like to know if it is possible to change this port so as not to conflict with other service that uses port 80?

Thank you

GuLinux, 2019-03-18:
Hi,
the software on its own doesn't use port `80`, it just provides a default `nginx` config file, but you can customise it or use your own entirely.
On a standard ubuntu installation, you can find this file in `/etc/nginx/sites-enabled/astrophotoplus.conf`.
I would recommend visiting nginx docs here: http://nginx.org/en/docs/

GuLinux, 2019-03-18:
Oh, I almost forgot: in theory, if you already have apache2, that will be used instead of nginx. You can just look for a similar file in apache `sites-enabled` directory.

Hope to be of help!
Cheers,
Marco

closed 2019-03-18 by GuLinux