GuLinux/AstroPhoto-Plus #140: phd2 connection

dragonlost, 2019-03-20:
Hello. 
is it expected to connect phd2 to allow dithering (with phd2 server connection) ?
 thank you.

GuLinux, 2019-03-20:
Hi,
not yet unfortunately.
I've just started implementing an autoguiding solution (without PHD2, but using [Donuts](https://github.com/jmccormac01/Donuts), a quite promising autoguiding algorithm), but it's still gonna be a while until I have a testable beta. It will of course also allow dithering.

I'm also thinking of allowing user to configure either the internal Donuts based autoguider, or PHD2 as an alternative, but if I'll do so it's gonna be a very simple integration (users will be expected to start and configure PHD2 on their own, AstroPhoto Plus will just connect and dither when necessary).

What are your thoughts on this? I might need an early beta tester when autoguiding will be implemented, would you be available for that?
Cheers,
Marco

dragonlost, 2019-03-20:
Ha super! That's exactly what I was hoping for!
I do not mind configuring phd2 by hand.

I am a developer on the NAFABox project. And we would like to integrate your software in!

dragonlost, 2019-03-20:
I am already beta tester for other software that integrates so no problem.

GuLinux, 2019-03-20:
Interesting project, I think we should really think on cooperating more :)
As you see I already tried to create a minimal ready-to-go raspberry pi image, so I guess we're heading in the same direction :)

I think I can get the PHD2 integration relatively quickly, possibly by the end of this week. It's gonna be just a simple control panel with `Connect/Disconnect`, `Start/Stop guiding`, possibly `Train`, and the dither options in the sequence job (`on/off`, `pixels`).

Sounds good?

dragonlost, 2019-03-20:
yes is very good !

The goal of NAFABox is to be a box for nomad astronomy. So your solution is ideal.
The EKOS Live solution is good too but is not suitable for use on nomad (without internet).

PR #166 (2019-09-30): Lightweight phd2 integration
Implement a small wrapper around PHD2 [EventMonitoring](https://github.com/OpenPHDGuiding/phd2/wiki/EventMonitoring) interface.

Minimum features:
 - [x] connection check at startup and/or session started
 - [x] notifications about socket connected/disconnected
 - [x] manual connection (could be replaced by automatically polling the server)
 - [x] report PHD2 status
 - [x] notifications for major events (connected, disconnected, star lost, guiding started, guiding stopped)
 - [x] dithering

Extra features to be implemented in #165 
closes #160 
fixes #140
closed 2019-09-30 by GuLinux