GuLinux/AstroPhoto-Plus #141: Autosetup issue on RPI

laheller, 2019-03-26:
**Describe the bug**
Following error comes when I call _sudo AstroPhotoPlus-ctl autosetup_:

```
Using nginx as web server
'/usr/lib/AstroPhotoPlus/frontend/webserver-conf/nginx/astrophotoplus.conf' -> '/etc/nginx/sites-available/astrophotoplus.conf'
Please enter the username for your AstroPhoto Plus backend server: pi

Failed to enable unit: Refusing to operate on linked unit file redis.service

Autoconfiguring pi as default AstroPhoto Plus user.
Run 'systemctl edit AstroPhotoPlus' to change user
Waiting for backend server to be ready............................................................................................................................................................................................
```

**To Reproduce**
Steps to reproduce the behavior:
1. Install the .deb package
2. Call autosetup 

**Expected behavior**
Backend created. It actually never creates, just the autosetup is waiting for backend server to be ready

**Screenshots**
N/A

**Server Information**
- AstroPhoto Plus Version: 1.0.1
 - Server hardware type: Raspberry
 - Operating system: Raspbian Stretch
 - Python version in use: ???

universe241981, 2019-03-27:
Hi,
The 
> Waiting for backend server to be ready.....

is actually the backend server installation still going on.
You can check the backend logs using the command `AstroPhotoPlus-ctl logs`.

laheller, 2019-03-28:
> Hi,
> The
> 
> > Waiting for backend server to be ready.....
> 
> is actually the backend server installation still going on.
> You can check the backend logs using the command `AstroPhotoPlus-ctl logs`.

OK, it is not an issue, I just had to wait + watch the logs.

closed 2019-03-28 by laheller

GuLinux, 2019-03-28:
Hi, sorry, busy couple of days..
Yes, there's still something going on with redis, but it's probably just a warning, not really an error, I'll look into it on the next patch release

laheller, 2019-04-20:
> 
> 
> Hi, sorry, busy couple of days..
> Yes, there's still something going on with redis, but it's probably just a warning, not really an error, I'll look into it on the next patch release

Nope, now the autosetup works.