GuLinux/AstroPhoto-Plus #142: camera exposure time value

hppetit, 2019-03-28:
Time exposure for a ZWO-ASI-120-MC camera jump from minimum to 1s. 
I can not select intermediate values

This behavior do not exist with Ekos and indi driver from AstroPhoto Plus

AstroPhoto Plus Version 1.0.1, Raspberry Pi 3 b+
Client browser: Chrome, Version 72.0.3626.121 (Build officiel) (64 bits)

Thank for your advice

GuLinux, 2019-03-28:
Hi,
I'm using the `step` value that comes from INDI.
Sub second exposures are really not our main target, but it should be possible to manually change the number to, say, 0.02.
Is this not the case?

hppetit, 2019-03-28:
I understand but for test purposes I tried to edit a value such as 0.02 but
without success.
Anyway, thank you very much for this wonderful project