GuLinux/AstroPhoto-Plus #143: hostapd service failed after update APP from b621 to b666

universe241981, 2019-03-29:
hostapd.service: Failed to load environment files: No such file or directory
Failed to start Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator

The `/etc/default/hostapd` -> `/usr/share/AstroPhotoPlus/config/raspberry_pi/hostapd`
But such file does not exist.

The same goes to `/etc/hostapd`

Device: RPi3b

universe241981, 2019-03-30:
I think the Raspberry image download link on the official website need to be update to the latest image with date 2019-03-18.
That will eliminate most of the problem and avoid issue from future user that try to update the image from b622 to b666.

GuLinux, 2020-06-18:
Closing, as with the new version hostapd is not used anymore, AstroPhoto Plus fully switched to NetworkManager

closed 2020-06-18 by GuLinux