GuLinux/AstroPhoto-Plus #144: Not really a bug but Installation unber Ubuntu 18.04 64bit on RPI3b+ fails

cstachon, 2019-04-13:
As I say not a bug as I doubt you support Ubuntu Mate 18.04 Beta 1 but might be useful :-)
Install fails and freezes RPI3b+ with message "Waiting for Back end server............." but I did notice that Dnsmasq failed to install earlier in the script.

This was a manual install after initial set up of RPI3B+ installed and updated - so I was using the "wget -N -q https://raw.githubusercontent.com/GuLinux/AstroPhoto-Plus/master/scripts/setup/ubuntu.sh && sudo bash ./ubuntu.sh && rm ubuntu.sh" but with SUDO

As I say FYI only

Good luck with an interesting project

universe241981, 2019-04-17:
Does the "......" continuously showing more dots?
Usually if under "Waiting for Back end server.............", the setup still running.

universe241981, 2019-04-17:
That's too short time for the wait.
Perhaps spare an hour for it to run. Although RPi3b+ not suppose to setup the backend at the long time frame.

mwasilew, 2019-12-03:
Not sure this is related but the setup script also failed for me:
```
./ubuntu.sh: line 68: python: command not found
--2019-12-03 22:22:16--  https://astrophotoplus.gulinux.net/development-builds/builds/latest/
Resolving astrophotoplus.gulinux.net (astrophotoplus.gulinux.net)... 178.62.102.250
Connecting to astrophotoplus.gulinux.net (astrophotoplus.gulinux.net)|178.62.102.250|:443... connected.
HTTP request sent, awaiting response... 403 Forbidden
2019-12-03 22:22:16 ERROR 403: Forbidden.

Reading package lists...
E: Unsupported file ./AstroPhotoPlus*.deb given on commandline
./ubuntu.sh: line 73: AstroPhotoPlus-ctl: command not found
```
I'm trying to debug it now. I'll share my findings or maybe propose a fix if it's easy.

GuLinux, 2020-03-30:
Hi, closing as this should now be properly fixed in recent build.

closed 2020-03-30 by GuLinux