GuLinux/AstroPhoto-Plus #145: [help request] Using raspberry pi camera

p1n9, 2019-06-07:
Hi all,

I'm trying to use the raspberry camera with AstroPhoto-Plus as a low cost imaging system.
I installed the AstroPhoto-Plus RPI img on a Raspberry 3B+, the sytem recognize well the pi cam (with v4l2 CCD driver), but I can't manage to get "real" image (or at least good).

With the default settings here is what I get : 
[image]
[https://imgur.com/2TSrQaw](https://imgur.com/2TSrQaw)

And this when I put the cap on the telescope :
[image]
[https://imgur.com/jtSCO7O](https://imgur.com/jtSCO7O)
So I gess this two pictures shows that something works somehow...

To prove that the picam sensor is healty, here is what I get with the [RPI-Cam-Web-Interface](https://elinux.org/RPi-Cam-Web-Interface) :
[image]
[https://imgur.com/2NpniZW](https://imgur.com/2NpniZW)
With default settings and on a pi-zero.

Also, perhaspe useful, I use the picam in "primefocus" (with the factory lens removed).

I hope someone of you could give me advises (or solution?!) on how to get my project to using AstroPhoto-Plus with the official raspberry camera fully working.

Again, images above was taken with default settings.
I've tried other things, like changing capture format (JFIF JPEG by default), but with any change I get a "capture error" message.
I'm not familiar to this things so, perhapse, I'm not doing it well.

I will pleased to give you more information if needed.

Cheers !

ps : I'm a newly noob in astrophotography and I'm seeking for some help

universe241981, 2019-06-11:
Hi, your case considered lucky with the picam working via the v4l2 CCD driver.
I've tried a not original picam which can be function using raspistill command on the RPi3b camera port. With v4l2 CCD driver in AstroPhoto-Plus, it show error as below:

> [ERROR] Error: unable to open device. Cannot identify /dev/video0: 2, No such file or directory

Anyway, the image that you taken with the v4l2 CCD driver in AstroPhoto-Plus is actually considered as normal? INDI driver didn't do color conversion on the images which it only capture RAW file which is in color bayer array form. That's why you got the pixel form image.
But if you are using the "Camera" tab in AstroPhoto-Plus for testing, it should show color image.

p1n9, 2019-06-11:
Hi,

The pixelated-grayscale image was taken from the Camera tab, and the "dark" shape can change regarding what I put in front of the camera.
Perhaps the v4l2 CCD driver of astrophoto-plus can't deal with a bare sensor like the RPI-Cam-Web-Interface do.
I will make more tests this week.

Concerning your error, have you loaded the v4l2 module ?
`sudo modeprobe bcm2835-v4l2`
(It should be installed by default, if not : `sudo apt-get install v4l-utils`)
You should now see the device video0 in /dev directory.
To load it at boot add a line with 'bcm2835-v4l2' in /etc/modules