GuLinux/AstroPhoto-Plus #149: Nothing seems to work

gerben-b, 2019-08-10:
I tested the install script on ubuntu 19.04 and on the new raspberry pi 4 (buster). On both I got an error about to many layers symlink.
I also tested ten image for the Raspberry pi and that did not do anything, also the wifi network dat was mentioned in the install sectio of the website did not come up
The scripted install resulted in a seemingly working website, but after that also nothing worked.

GuLinux, 2019-08-11:
Hi, that's a weird error, could you try to get a full log attached please?
Cheers,
Marco

gerben-b, 2019-08-12:
Hi Marco,

I don't have that anymore, I cleaned it up afterwards, but still I thought this could be worth your while: 
- A scripted install on Ubuntu 19.04 failed (too many symlinks);
- A scripted install on raspberry (buster) failed for the same reason;
- The image recommended based install on raspberry failed (no wifi).
Ubuntu 10.04 and Debian Buster may be too new and the Raspberry 4 may also be too new.

Could it be that your scripts and image only work on (specific) older versions?

Cheers,
Gerben

gerben-b, 2019-08-12:
I just tried it on ubuntu 18.04 and got this
Failed to enable unit: Refusing to operate on linked unit file redis.service
Autoconfiguring gerben as default AstroPhoto Plus user.
Run 'systemctl edit AstroPhotoPlus' to change user
Created symlink /etc/systemd/system/multi-user.target.wants/AstroPhotoPlus.service → /usr/lib/systemd/system/AstroPhotoPlus.service.
Waiting for backend server to be ready...........................................................................................................................................................................................................................................................

This time no symlink error, but a redis error instead. Also it waits indefinitely for the backend to be ready. Did you assume some requirements that are not mentioned in the description on the site?

Cheers,
Gerben

universe241981, 2019-08-13:
Hi,
The last time I use AstroPhoto-Plus is of the release on March 18, 2019 with b666 on the file name.
This b666 image works fine on RPi3b with Raspbian Stretch as base.
Perhaps you can download the b666 image for ubuntu and try? Or you already try that?

gerben-b, 2019-09-02:
That takes too much of my time and I prefer to use the latest version. If I use an older one that works, then I won't be able to update.

disablewong, 2019-11-28:
I got this funny error too, which mentioned the too many layer of symlinks.
This error actually came with redis-server.service but after checking systemctl status redis-server.service, it say redis-service was active.

However, I never able to start the AstroPhoto-Plus service active in systemd.
I've also got "Waiting for backend server to be ready..........................................................................................................................................................................................................................................................."
On browser localhost, it says the backend server was offline. 

So i suspected it's a bug between the redis and AstroPhoto-Plus inside the script, but I couldn't reveal where has gone wrong.  
I had tried it on RK3399 and 6818 SBC (Both are nanopi) with ubuntu and debian buster, install with deb/tar.gz all combinations got the same result..... 
There was not much information about compilation so I couldn't do it.......

May I suggest there be some compilation instructions together with the dependencies included in the README.md inside the tarball so desperate people could try to do a compilation from scratch.... like what I did for Planetary Imager ;)

Anayways, thanks for your great works for both softwares Marco!
cheers,

Vasco

GuLinux, 2019-11-28:
Hi,
sorry this seems to be inactive. I've actually been working on the project recently, mainly changing the CI/CD system and integrating PHD2, so the project is well alive, but time is always a very scarce resource.

Firstly, I don't know exactly what you refer to when you ask about "compilation instructions". There really isn't much to compile, a good part of the project is in python3, the rest in javascript (if you grab the tarball from the release page it's already transpiled for you, so no further steps are required).
There's a small amount of py-c++ modules that need to be compiled, but that's handled automatically by the `start-server` script (which is the one invoked by systemd).

Anyway, I did encounter the main issue as well recently on a raspberry pi, and yes, I believe there might be some weird interaction between AstroPhoto Plus and redis. In my case, restarting redis first, and AstroPhoto Plus subsequently seemed to temporarily solve the issue.

I don't really have a quick solution at the moment. There is a connection retry routine in AstroPhoto Plus, pinging redis, but I think it gives up after a few retries, so a possible workaround could be to increase the attempts number, or possibly retry indefinitely. There might also be some issue in detecting whether redis is really up. I'll look into that as soon as possible.

A more definitive solution might be to just ditch redis in favour of something else; this is already planned in #154 .
It is of course a much bigger task, so I really can't offer lots of hopes at the moment, but please try restarting the services and let me know if it actually helps.
Cheers,
Marco

gerben-b, 2019-11-30:
Hi Marco,
I tried the easy install and followed the instructions. In the browser i got "internal server error" and no information in the error log
Also updating ubuntu now has errors with indi.
Any ideas?
Kind regards,Gerben
Outlook voor Android downloaden

PR #183 (2020-03-19): Start redis together with application as a managed service.
Closes #182 
Closes #149
commit dd5f48bc (2020-03-19): Start redis together with application as a managed service. (#183)

Closes #182
Closes #149
closed 2020-03-19 by GuLinux