GuLinux/AstroPhoto-Plus #177: Error when running AstroPhotoPlus on RPi4 (latest Raspbian 2020-02-05)

aruangra, 2020-02-14:
**Describe the bug**
When starting the INDI server, the webpage shows:...
"Offline
It looks like the backend server is not running.
The application will reload as soon as connection will be established again.
If that doesn't happen, please either check your network connection, that the server is connected and running, and server logs using the following command:"

I confirm that the INDI server could be started using ps axw
30012 ?        S      0:00 /usr/bin/indiserver /usr/bin/indi_simulator_ccd
30014 ?        S      0:00 /usr/bin/indi_simulator_ccd

**To Reproduce**
Steps to reproduce the behavior:
I install ngrix and INDI first and then install the package using...

sudo apt-get install -y ./AstroPhotoPlus-1.0.1-b850-Raspbian.deb

sudo nano /usr/bin/AstroPhotoPlus-ctl
**************** edited due to an error **************************
start-redis() {
    systemctl enable /lib/systemd/system/redis-server.service
    is-chroot || systemctl start /lib/systemd/system/redis-server.service
}
************************************************

sudo AstroPhotoPlus-ctl autosetup pi

**Expected behavior**
INDI starts and ready for operating devices

**Screenshots**
I add the logs file here
https://drive.google.com/file/d/1h2RWT0no7hZrCc8Tz5pA1BGzlnIAqDOZ/view?usp=sharing

When running sudo AstroPhotoPlus-ctl autosetup pi, the screen shows:

pi@raspberrypi:~ $ sudo AstroPhotoPlus-ctl autosetup pi
Using nginx as web server
'/usr/lib/AstroPhotoPlus/frontend/webserver-conf/nginx/astrophotoplus.conf' -> '/etc/nginx/sites-available/astrophotoplus.conf'
Failed to start lib-systemd-system-redis\x2dserver.service.mount: Unit lib-systemd-system-redis\x2dserver.service.mount not found.
Autoconfiguring pi as default AstroPhoto Plus user.
Run 'systemctl edit AstroPhotoPlus' to change user
Waiting for backend server to be ready.
Backend ready. You can now start your browser.
Automatic setup complete. You can open your browser at http://localhost to start using your AstroPhoto Plus installation.
If this is the first time installing AstroPhoto Plus, the server might not be ready yet and display an error page.
If this is the case, run /usr/bin/AstroPhotoPlus-ctl logs to see if it's still installing packages, or if there is a problem with your setup.

**Server Information**
- AstroPhoto Plus Version
AstroPhotoPlus-1.0.1-b850-Raspbian.deb
 - Server hardware type (e.g. generic x86 or x86_64 computer, Raspberry, other ARM devices, Intel Compute Stick)
RPi4 4GB
 - Operating system (e.g. Ubuntu Linux 18.04)
Raspbian as of 2002-02-05
 - Python version in use
Python 3.7.3 
 - Remember to attach server logs created with `AstroPhotoPlus-ctl logs` if applicable.
https://drive.google.com/file/d/1h2RWT0no7hZrCc8Tz5pA1BGzlnIAqDOZ/view?usp=sharing

**Client**
 - Device: [e.g. iPhone6, desktop]
PC desktop
 - OS: [e.g. iOS8.1, Windows, Linux]
Windows 10
 - Browser [e.g. Chrome, Firefox]
Chrome

**Additional context**
The complete steps from the  fresh raspbian image

1. Initial setup

sudo raspi-config
set resolution
enable ssh, vnc

sudo apt-get update
sudo apt-get upgrade

sudo apt-get install nano aptitude gcc g++ make usbutils bc wget build-essential bc

sudo apt-get install network-manager

2. Web server

sudo apt-get install nginx php7.3-fpm php7.3-cgi php7.3-cli php7.3-common php7.3-xml

8. INDI

REFERENCE: 
https://github.com/indilib/indi
https://github.com/indilib/indi-3rdparty

sudo apt-get install libnova-dev libcfitsio-dev libusb-1.0-0-dev zlib1g-dev libgsl-dev build-essential cmake git libjpeg-dev libcurl4-gnutls-dev libtiff-dev libfftw3-dev libftdi-dev libgps-dev libraw-dev libdc1394-22-dev libgphoto2-dev libboost-dev libboost-regex-dev librtlsdr-dev liblimesuite-dev libftdi1-dev libavcodec-dev libavdevice-dev

****** INDI CORE 

cd /home/pi/
mkdir -p ~/Projects
cd ~/Projects
git clone --depth 1 https://github.com/indilib/indi.git

mkdir -p ~/Projects/build/indi-core
cd ~/Projects/build/indi-core
cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ~/Projects/indi
make -j4
sudo make install

****** INDI 3rd Party

cd ~/Projects
git clone --depth=1 https://github.com/indilib/indi-3rdparty

mkdir -p ~/Projects/build/indi-3rdparty-libs
cd ~/Projects/build/indi-3rdparty-libs
cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug -DBUILD_LIBS=1 ~/Projects/indi-3rdparty
make -j4
sudo make install

mkdir -p ~/Projects/build/indi-3rdparty
cd ~/Projects/build/indi-3rdparty
cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ~/Projects/indi-3rdparty
make -j4
sudo make install

12. AstroPhotoPlus
Reference: https://github.com/GuLinux/AstroPhoto-Plus/releases

cd /home/pi/

sudo apt-get install -y ./AstroPhotoPlus-1.0.1-b850-Raspbian.deb

sudo nano /usr/bin/AstroPhotoPlus-ctl
************************************************
start-redis() {
    systemctl enable /lib/systemd/system/redis-server.service
    is-chroot || systemctl start /lib/systemd/system/redis-server.service
}
************************************************

sudo AstroPhotoPlus-ctl autosetup pi

**************** ERROR after starting INDI server****************************

Offline
It looks like the backend server is not running.

The application will reload as soon as connection will be established again.

If that doesn't happen, please either check your network connection, that the server is connected and running, and server logs using the following command:

AstroPhotoPlus-ctl logs

GuLinux, 2020-02-14:
Hi,
The current setup is a bit unstable, particularly on Raspberry Pi.
It might be due to Redis playing a bit, I would suggest as a workaround:
 - restart redis
`sudo systemctl restart redis` (or `redis-server`)
 - restart AstroPhoto Plus
`sudo AstroPhotoPlus-ctl  restart`

I will be working on removing the redis dependency entirely when possible (see #154 ), I'll also see if I can do something as a short term solution as soon as possible, as that might be a big task

aruangra, 2020-02-14:
Thank you for the workaround.

GuLinux, 2020-03-30:
Hi, closing as this should now be properly fixed in recent build.

closed 2020-03-30 by GuLinux