GuLinux/AstroPhoto-Plus #197: backend installation

ettoreribaudo, 2020-05-15:
pi@raspberrypi:~ $ AstroPhotoPlus-ctl logs
-- Logs begin at Thu 2019-02-14 11:11:59 CET. --
mag 15 20:18:11 raspberrypi systemd[1]: Starting Astrophotography sequence server...
-- Subject: L'unità AstroPhotoPlus.service inizia la fase di avvio
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- L'unità AstroPhotoPlus.service ha iniziato la fase di avvio.
mag 15 20:18:12 raspberrypi systemd[1]: Started Astrophotography sequence server.
-- Subject: L'unità AstroPhotoPlus.service termina la fase di avvio
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- L'unità AstroPhotoPlus.service ha terminato la fase di avvio.
-- 
-- La fase di avvio è done.
mag 15 20:18:33 raspberrypi AstroPhoto-Plus[419]: Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
mag 15 20:18:33 raspberrypi AstroPhoto-Plus[419]: Requirement already satisfied: wheel in /usr/lib/python3/dist-packages (0.32.3)
mag 15 20:18:37 raspberrypi AstroPhoto-Plus[419]: Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
mag 15 20:18:37 raspberrypi AstroPhoto-Plus[419]: Collecting pyindi-client>=0.2.3 (from -r requirements.txt (line 1))
mag 15 20:18:37 raspberrypi AstroPhoto-Plus[419]:   Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0xb526cc70>: Failed to establish a new connection: [Errno -3] Errore temporaneo nella risoluzione del nome')': /simple/pyindi-client/
mag 15 20:18:38 raspberrypi AstroPhoto-Plus[419]:   Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0xb526cd10>: Failed to establish a new connection: [Errno -3] Errore temporaneo nella risoluzione del nome')': /simple/pyindi-client/
mag 15 20:18:39 raspberrypi AstroPhoto-Plus[419]:   Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0xb526c430>: Failed to establish a new connection: [Errno -3] Errore temporaneo nella risoluzione del nome')': /simple/pyindi-client/
mag 15 20:18:41 raspberrypi AstroPhoto-Plus[419]:   Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0xb526c9b0>: Failed to establish a new connection: [Errno -3] Errore temporaneo nella risoluzione del nome')': /simple/pyindi-client/
mag 15 20:18:45 raspberrypi AstroPhoto-Plus[419]:   Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0xb4fb5ab0>: Failed to establish a new connection: [Errno -3] Errore temporaneo nella risoluzione del nome')': /simple/pyindi-client/
mag 15 20:18:45 raspberrypi AstroPhoto-Plus[419]:   Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0xb4fb5b10>: Failed to establish a new connection: [Errno -3] Errore temporaneo nella risoluzione del nome')': /simple/pyindi-client/
mag 15 20:18:45 raspberrypi AstroPhoto-Plus[419]:   Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0xb4fb5f70>: Failed to establish a new connection: [Errno -3] Errore temporaneo nella risoluzione del nome')': /simple/pyindi-client/
mag 15 20:18:46 raspberrypi AstroPhoto-Plus[419]:   Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0xb4fb5850>: Failed to establish a new connection: [Errno -3] Errore temporaneo nella risoluzione del nome')': /simple/pyindi-client/
mag 15 20:18:48 raspberrypi AstroPhoto-Plus[419]:   Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0xb527c030>: Failed to establish a new connection: [Errno -3] Errore temporaneo nella risoluzione del nome')': /simple/pyindi-client/
mag 15 20:18:52 raspberrypi AstroPhoto-Plus[419]:   Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0xb527cf90>: Failed to establish a new connection: [Errno -3] Errore temporaneo nella risoluzione del nome')': /simple/pyindi-client/
mag 15 20:18:52 raspberrypi AstroPhoto-Plus[419]:   Could not find a version that satisfies the requirement pyindi-client>=0.2.3 (from -r requirements.txt (line 1)) (from versions: )
mag 15 20:18:52 raspberrypi AstroPhoto-Plus[419]: No matching distribution found for pyindi-client>=0.2.3 (from -r requirements.txt (line 1))
mag 15 20:18:52 raspberrypi systemd[1]: AstroPhotoPlus.service: Main process exited, code=exited, status=1/FAILURE
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- An ExecStart= process belonging to unit AstroPhotoPlus.service has exited.
-- 
-- The process' exit code is 'exited' and its exit status is 1.
mag 15 20:18:52 raspberrypi systemd[1]: AstroPhotoPlus.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- The unit AstroPhotoPlus.service has entered the 'failed' state with result 'exit-code'.

GuLinux, 2020-05-15:
Sorry, without context it's difficult to diagnose anything.
I'd need a lot more information, like Raspbian version (if it's too old it might not be supported), if it's installed from scratch or a new version, etc.

A few notable things though:
```
-- Logs begin at Thu 2019-02-14 11:11:59 CET. --
```
The date is wrong here. This is a first clue, then
```
mag 15 20:18:45 raspberrypi AstroPhoto-Plus[419]:   Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0xb4fb5f70>: Failed to establish a new connection: [Errno -3] Errore temporaneo nella risoluzione del nome')': /simple/pyindi-client/
```
Looks like your raspberry pi is not connected to internet. Which is probably also the reason for the wrong date (Raspberry Pi don't have a hardware RTC, they just download the time from Internet).

AstroPhoto Plus needs an internet connection to complete setup, please ensure your raspberry pi can access the web.
I'm going to close the issue (feel free to keep commenting though, or reopening if something else is wrong).

closed 2020-05-15 by GuLinux

ettoreribaudo, 2020-05-15:
So Raspian for 4b, but I resolved, my connections are randomic.
Regards
Ettore

GuLinux, 2020-05-15:
Good to hear :) I hope you'll find the app useful :)
Cheers,
Marco