GuLinux/AstroPhoto-Plus #209: canon 1300d busy

ettoreribaudo, 2020-05-19:
I have a problem with canon 1300d, it is recognized but everything remains in busy.
[image]

GuLinux, 2020-05-19:
Hi,
This looks like an INDI problem, and apparently there's already a very similar issue raised in INDI forum: https://indilib.org/forum/ccds-dslrs/4352-canon-dslr-stopped-working-incomplete-connection.html?start=0

I'll see if I can replicate it on my cameras and investigate (I actually never tried using DSLR cameras, so it might be an interesting experiment).

In the meantime, I would suggest, if you can, to try and install KStars/Ekos on that very same machine (Raspberry Pi I think?). If you still see the same behaviour, then it's definitely an INDI problem.

Cheers

ettoreribaudo, 2020-05-19:
I installed astroberry, everything works perfectly with Ekos.

ettoreribaudo, 2020-05-19:
Is it possible that in AstroPhoto the canon 1300D wants an SD card to work?

GuLinux, 2020-05-19:
Hi,
AstroPhoto Plus doesn't _directly_ manage any hardware, everything is delegated to INDI (same as KStars and Astroberry). This is why I suspect there is an underlying issue. Since you say everything seems to work fine, I suspect there might just be a different setting between the two softwares. I'll have to investigate.

Same applies to the SD card question: AstroPhoto Plus per se shouldn't require it. It's usually a camera setting, and by default Canon cameras require the SD Card, so I'd probably check if this option is enabled for you.

ettoreribaudo, 2020-05-19:
OK I check

ettoreribaudo, 2020-05-19:
I verified but nothing,this is a error:
k_manager.py active-connections --json
mag 19 17:38:34 raspberrypi sudo[21622]: pam_unix(sudo:session): session opened for user root by (uid=0)
mag 19 17:38:34 raspberrypi sudo[21622]: pam_unix(sudo:session): session closed for user root
mag 19 17:38:35 raspberrypi sudo[21662]:       pi : TTY=unknown ; PWD=/usr/lib/AstroPhotoPlus/backend ; USER=root ; COMMAND=/usr/lib/AstroPhotoPlus/backend/network/network_manager.py connections --json
mag 19 17:38:35 raspberrypi sudo[21662]: pam_unix(sudo:session): session opened for user root by (uid=0)
mag 19 17:38:35 raspberrypi AstroPhoto-Plus[18372]: Dispatch command error(-1): INDI: Could not find property CCD_EXPOSURE in Canon DSLR EOS 1300D
mag 19 17:38:35 raspberrypi AstroPhoto-Plus[18372]: <setNumberVector device="Canon DSLR EOS 1300D" name="CCD_EXPOSURE" state="Idle" timeout="60" timestamp="2020-05-19T15:38:35">
mag 19 17:38:35 raspberrypi AstroPhoto-Plus[18372]:     <oneNumber name="CCD_EXPOSURE_VALUE" min="0.00025" max="3600" step="1">
mag 19 17:38:35 raspberrypi AstroPhoto-Plus[18372]: 1
mag 19 17:38:35 raspberrypi AstroPhoto-Plus[18372]:     </oneNumber>
mag 19 17:38:35 raspberrypi AstroPhoto-Plus[18372]: </setNumberVector>
mag 19 17:38:35 raspberrypi AstroPhoto-Plus[18372]: Dispatch command error(-1): INDI: Could not find property CCD_EXPOSURE in Canon DSLR EOS 1300D
mag 19 17:38:35 raspberrypi AstroPhoto-Plus[18372]: <setNumberVector device="Canon DSLR EOS 1300D" name="CCD_EXPOSURE" state="Idle" timeout="60" timestamp="2020-05-19T15:38:35">
mag 19 17:38:35 raspberrypi AstroPhoto-Plus[18372]:     <oneNumber name="CCD_EXPOSURE_VALUE" min="0.00025" max="3600" step="1">
mag 19 17:38:35 raspberrypi AstroPhoto-Plus[18372]: 1
mag 19 17:38:35 raspberrypi AstroPhoto-Plus[18372]:     </oneNumber>
mag 19 17:38:35 raspberrypi AstroPhoto-Plus[18372]: </setNumberVector>
mag 19 17:38:36 raspberrypi sudo[21662]: pam_unix(sudo:session): session closed for user root
mag 19 17:38:36 raspberrypi sudo[21704]:       pi : TTY=unknown ; PWD=/usr/lib/AstroPhotoPlus/backend ; USER=root ; COMMAND=/usr/lib/AstroPhotoPlus/backend/network/network_manager.py active-connections --json
mag 19 17:38:36 raspberrypi sudo[21704]: pam_unix(sudo:session): session opened for user root by (uid=0)
mag 19 17:38:36 raspberrypi sudo[21704]: pam_unix(sudo:session): session closed for user root
mag 19 17:38:37 raspberrypi sudo[21751]:       pi : TTY=unknown ; PWD=/usr/lib/AstroPhotoPlus/backend ; USER=root ; COMMAND=/usr/lib/AstroPhotoPlus/backend/network/network_manager.py connections --json
mag 19 17:38:37 raspberrypi sudo[21751]: pam_unix(sudo:session): session opened for user root by (uid=0)
mag 19 17:38:38 raspberrypi sudo[21751]: pam_unix(sudo:session): session closed for user root
mag 19 17:38:38 raspberrypi sudo[21760]:       pi : TTY=unknown ; PWD=/usr/lib/AstroPhotoPlus/backend ; USER=root ; COMMAND=/usr/lib/AstroPhotoPlus/backend/network/net

GuLinux, 2020-05-19:
Hi,
I was able to reproduce the problem with my camera.

In the end it was indeed an INDI issue. Oddly enough, there seems to be an INDI property with no name or label. KStars/Ekos added a workaround to simply ignore the property, which is what I've done myself in the current build.

Please try out next build (it should be 1.0.1-b893 or above) when it gets automatically generated.

**note** there is also another step before actually being able to capture images. You need to find your camera information (image resolution and pixel pitch) and add the corresponding info in the "Image Info" section. I think you need to add the resolution in "Image Settings" too.

I will try implementing a more user friendly procedure in the future.
[image]

closed 2020-05-19 by GuLinux

ettoreribaudo, 2020-05-19:
Thank Marco, i reinstalled everything, the Canon 1300d works, but my wifi has random disconnection problems

GuLinux, 2020-05-20:
I see..
I'm afraid for WiFi I can't do much, WiFi setup is mainly done by the OS (NetworkManager), in AstroPhoto Plus we only create/activate connections.

Are you talking about Access Point mode, or Station (client)?
I guess in the first case, one thing could be to be able to choose the WiFi channel. You can actually edit the connection using NetworkManager and see if that helps. Perhaps I could add a "channel selection" to AstroPhoto Plus.
If it's in station mode, I probably really can't do much

commit 21fc9303 (2020-05-19): Fix initialisation error on INDI Property with no name (GPhoto devices - Canon) (#210)

fixes #209