GuLinux/AstroPhoto-Plus #213: Astrometry

ettoreribaudo, 2020-05-20:
Why not upload catalogs directly to sd for astrometry?

GuLinux, 2020-05-20:
Hi,
Sorry, what do you mean exactly? Do you mean the astrometry index files, or the NGC/IC/Messier database?

And by "directly to sd", do you mean as in "user can download manually and save them into the SD"?

ettoreribaudo, 2020-05-20:
> Hi,
> Sorry, what do you mean exactly? Do you mean the astrometry index files, or the NGC/IC/Messier database?
> 
> And by "directly to sd", do you mean as in "user can download manually and save them into the SD"?

Sorry, what do you mean exactly? Do you mean the astrometry index files, or the NGC/IC/Messier database? **Astrometry for plate solve**r

And by "directly to sd", do you mean as in "user can download manually and save them into the SD"? **in directory in to Sd Card**

GuLinux, 2020-05-21:
I guess I could add a setting to change the location of plate solving index files, is that what you meant?

ettoreribaudo, 2020-05-21:
I mean if it is possible to load the Ucac3 catalog in a resident way

GuLinux, 2020-05-21:
Yes, possibly (I need to look up where to find it), but that's a different things from astrometry indexes though

GuLinux, 2020-06-18:
I'll close this for now, as extra catalog can now be accessed through SIMBAD ( #235 )

closed 2020-06-18 by GuLinux