GuLinux/AstroPhoto-Plus #222: Make Indi install optional

tkottary, 2020-05-29:
**Is your feature request related to a problem? Please describe.**
I already have indi nightly installed and do not want to install the stable version

**Describe the solution you'd like**
give option to skip indi installation 

**Describe alternatives you've considered**
re-install indi nightly after astrophotoplus is installed.

**Additional context**
Add any other context or screenshots about the feature request here.

GuLinux, 2020-06-01:
Hi,
sorry, do you mean in the "easy install" script?
If that's the case, I'd rather leaving it as it is, as I feel the aim should be to ask as less questions as possible, and INDI per se is not really an "optional" component to which a regular user would likely say "no".

For advanced installation, there's always the option of installing the `.deb` package manually, which only depends on `libindi-dev` (which I guess should be available in INDI nightly as well) and then manually running `AstroPhotoPlus-ctl autosetup`

GuLinux, 2020-06-01:
(closing unless clarified otherwise)

closed 2020-06-01 by GuLinux