GuLinux/AstroPhoto-Plus #225: Canon 1300D pixel

ettoreribaudo, 2020-05-31:
There are some errors, the Canon has 14 bits with 5184 x 3456 4.3 pixels, but the data is wrong and any value that I put them back is wrong.
[image]
[image]

ettoreribaudo, 2020-05-31:
Now the log gives me this error, after update ver 911 with Canon 
Aug 31 18:53:50 raspberrypi sudo [3944]: pam_unix (sudo: session): session closed for user root
May 31 18:53:50 raspberrypi sudo [3954]: pi: TTY = unknown; PWD = / usr / lib / AstroPhotoPlus / backend; USER = root; COMMAND = / usr / lib / AstroPhotoPlus / backend / network / network_manager.py active-connections --json
May 31 18:53:50 raspberrypi sudo [3954]: pam_unix (sudo: session): session opened for user root by (uid = 0)
May 31 18:53:50 raspberrypi sudo [3954]: pam_unix (sudo: session): session closed for user root
May 31 18:53:51 raspberrypi AstroPhoto-Plus [460]: INDI server localhost / 7624 disconnected.
May 31 18:53:51 raspberrypi AstroPhoto-Plus [460]: INDI server localhost / 7624 disconnected.
May 31 18:53:51 raspberrypi AstroPhoto-Plus [460]: [2020-05-31 18:53:51 +0200] [626] [WARNING] BLOBClient: Unexpected disconnection from INDI server with error code -1, trying to connect
May 31 18:53:51 raspberrypi sudo [3979]: pi: TTY = unknown; PWD = / usr / lib / AstroPhotoPlus / backend; USER = root; COMMAND = / usr / lib / AstroPhotoPlus / backend / network / network_manager.py connections --json
May 31 18:53:51 raspberrypi sudo [3979]: pam_unix (sudo: session): session opened for user root by (uid = 0)
May 31 18:53:51 raspberrypi sudo [3979]: pam_unix (sudo: session): session closed for user root
May 31 18:53:51 raspberrypi sudo [3990]:

ettoreribaudo, 2020-06-01:
I resolved see the canon but not bit and pixel.

GuLinux, 2020-06-01:
Hi,
this is already planned as #211 .
INDI doesn't seem to be capable of getting DSLR information, there isn't much I can do on AstroPhoto Plus, except implementing a wizard asking the user to manually store camera information (which is what #211 is about).

An alternative could also be to implement a "Camera database" with all these infos, but I don't see this as a viable alternative in the near future.

closed 2020-06-01 by GuLinux