GuLinux/AstroPhoto-Plus #239: Versione 922

ettoreribaudo, 2020-06-18:
After updating to version 922, nothing works, not even INDI
[image]
[image]

GuLinux, 2020-06-18:
That can happen if the backend server is not replying. I need to see some backend logs.
Have you tried just rebooting the server?

ettoreribaudo, 2020-06-18:
- Logs begin at Thu 2020-06-18 13:49:51 BST. --
Jun 18 13:49:55 AstroPhoto-Plus systemd[1]: Starting Astrophotography sequence server...
-- Subject: A start job for unit AstroPhotoPlus.service has begun execution
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- A start job for unit AstroPhotoPlus.service has begun execution.
-- 
-- The job identifier is 92.
Jun 18 13:49:55 AstroPhoto-Plus systemd[1]: Started Astrophotography sequence server.
-- Subject: A start job for unit AstroPhotoPlus.service has finished successfully
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- A start job for unit AstroPhotoPlus.service has finished successfully.
-- 
-- The job identifier is 92.
Jun 18 13:49:56 AstroPhoto-Plus AstroPhoto-Plus[437]: + GUNICORN_BINARY=gunicorn3
Jun 18 13:49:56 AstroPhoto-Plus AstroPhoto-Plus[437]: + which gunicorn3
Jun 18 13:49:56 AstroPhoto-Plus AstroPhoto-Plus[437]: + GUNICORN_BINARY=gunicorn
Jun 18 13:49:56 AstroPhoto-Plus AstroPhoto-Plus[437]: + gunicorn --threads 4 -t 30 -k gthread -b 0.0.0.0:5000 api_entrypoint:app --log-level warning --log-syslog --log-syslog-prefix AstroPhotoPlus --disable-redirect-access-to-syslog
Jun 18 13:49:57 AstroPhoto-Plus AstroPhoto-Plus[437]: **** Initialising Redis server on port 45625
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]: [2020-06-18 13:50:02 +0100] [593] [ERROR] Exception in worker process
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]: Traceback (most recent call last):
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]:   File "/home/stargazer/.local/share/AstroPhotoPlus/python-venv/lib/python3.7/site-packages/gunicorn/arbiter.py", line 583, in spawn_worker
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]:     worker.init_process()
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]:   File "/home/stargazer/.local/share/AstroPhotoPlus/python-venv/lib/python3.7/site-packages/gunicorn/workers/gthread.py", line 92, in init_process
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]:     super().init_process()
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]:   File "/home/stargazer/.local/share/AstroPhotoPlus/python-venv/lib/python3.7/site-packages/gunicorn/workers/base.py", line 119, in init_process
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]:     self.load_wsgi()
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]:   File "/home/stargazer/.local/share/AstroPhotoPlus/python-venv/lib/python3.7/site-packages/gunicorn/workers/base.py", line 144, in load_wsgi
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]:     self.wsgi = self.app.wsgi()
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]:   File "/home/stargazer/.local/share/AstroPhotoPlus/python-venv/lib/python3.7/site-packages/gunicorn/app/base.py", line 67, in wsgi
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]:     self.callable = self.load()
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]:   File "/home/stargazer/.local/share/AstroPhotoPlus/python-venv/lib/python3.7/site-packages/gunicorn/app/wsgiapp.py", line 49, in load
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]:     return self.load_wsgiapp()
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]:   File "/home/stargazer/.local/share/AstroPhotoPlus/python-venv/lib/python3.7/site-packages/gunicorn/app/wsgiapp.py", line 39, in load_wsgiapp
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]:     return util.import_app(self.app_uri)
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]:   File "/home/stargazer/.local/share/AstroPhotoPlus/python-venv/lib/python3.7/site-packages/gunicorn/util.py", line 358, in import_app
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]:     mod = importlib.import_module(module)
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]:   File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]:     return _bootstrap._gcd_import(name[level:], package, level)
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]:   File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]:   File "<frozen importlib._bootstrap>", line 983, in _find_and_load
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]:   File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]:   File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]:   File "<frozen importlib._bootstrap_external>", line 728, in exec_module
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]:   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]:   File "/usr/lib/AstroPhotoPlus/backend/api_entrypoint.py", line 2, in <module>
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]:     from app import app
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]:   File "/usr/lib/AstroPhotoPlus/backend/app.py", line 35, in <module>
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]:     from broadcast_service import broadcast_service
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]:   File "/usr/lib/AstroPhotoPlus/backend/broadcast_service.py", line 4, in <module>
Jun 18 13:50:02 AstroPhoto-Plus AstroPhoto-Plus[437]:     from system import settings, syslog
[... cut ...]
Jun 18 13:50:44 AstroPhoto-Plus AstroPhoto-Plus[437]:     from .platesolving import PlateSolving
Jun 18 13:50:44 AstroPhoto-Plus AstroPhoto-Plus[437]:   File "/usr/lib/AstroPhotoPlus/backend/platesolving/platesolving.py", line 17, in <module>
Jun 18 13:50:44 AstroPhoto-Plus AstroPhoto-Plus[437]:     from PIL import Image, UnidentifiedImageError
Jun 18 13:50:44 AstroPhoto-Plus AstroPhoto-Plus[437]: ImportError: cannot import name 'UnidentifiedImageError' from 'PIL' (/usr/lib/python3/dist-packages/PIL/__init__.py)

ettoreribaudo, 2020-06-18:
I remove all and reinstall but nothing same error.
Ettore

ettoreribaudo, 2020-06-18:
Only version 922 have problem.
I have installed only 2020-06-07-AstroPhoto-Plus-1.0.1-b916-RaspberryPi.zip but nothing same error and block.

GuLinux, 2020-06-18:
Hi,
looks like it was a simple dependency issue. I've updated the `requirements.txt` file, that should fix it (I tested on Ubuntu Bionic to be sure).

Please have a look as soon as the next release will be built and published ( >= 923)

commit 9f52faee (2020-06-18): Require PILLOW v>=7.0.0; fixes #239
closed 2020-06-18 by GuLinux