GuLinux/AstroPhoto-Plus #240: Goto

ettoreribaudo, 2020-06-18:
Version 923 is ok. I tried GOTO but my mount not move to object.
[image]
[image]
[image]

GuLinux, 2020-06-18:
You might want to check all indi properties (is the mount parked? Is it tracking? Is there any error in the indi telescope page?).
If the behaviour persists please open another ticket though, and most importantly, also include full backend logs (in debug mode).

ettoreribaudo, 2020-06-18:
Mount is unparked and tracking, nothing error in telescope page.
Ok i try with debug.

ettoreribaudo, 2020-06-18:
[indi_pmc8_telescope_18:12:29.log](https://github.com/GuLinux/AstroPhoto-Plus/files/4800485/indi_pmc8_telescope_18.12.29.log)

GuLinux, 2020-06-18:
Sorry, I meant putting AstroPhoto Plus logs in debug mode, and sending me
the backend server logs. That looks like an INDI device log.

One last thing though: did you check whether the telescope coordinates were
actually set to the target?

ettoreribaudo, 2020-06-19:
Sorry for delay 
-- Logs begin at Thu 2020-06-18 20:37:31 CEST. --
giu 18 20:37:35 raspberrypi systemd[1]: Starting Astrophotography sequence server...
-- Subject: L'unità AstroPhotoPlus.service inizia la fase di avvio
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- L'unità AstroPhotoPlus.service ha iniziato la fase di avvio.
giu 18 20:37:35 raspberrypi systemd[1]: Started Astrophotography sequence server.
-- Subject: L'unità AstroPhotoPlus.service termina la fase di avvio
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- L'unità AstroPhotoPlus.service ha terminato la fase di avvio.
-- 
-- La fase di avvio è done.
giu 18 20:37:36 raspberrypi AstroPhoto-Plus[445]: + GUNICORN_BINARY=gunicorn3
giu 18 20:37:36 raspberrypi AstroPhoto-Plus[445]: + which gunicorn3
giu 18 20:37:36 raspberrypi AstroPhoto-Plus[445]: + GUNICORN_BINARY=gunicorn
giu 18 20:37:36 raspberrypi AstroPhoto-Plus[445]: + gunicorn --threads 4 -t 30 -k gthread -b 0.0.0.0:5000 api_entrypoint:app --log-level warning --log-syslog --log-syslog-prefix AstroPhotoPlus --disable-redirect-access-to-syslog
giu 18 20:37:37 raspberrypi AstroPhoto-Plus[445]: **** Initialising Redis server on port 47085
giu 18 20:37:45 raspberrypi AstroPhoto-Plus[445]: setting log level from WARNING to WARNING
giu 19 12:40:07 raspberrypi AstroPhoto-Plus[445]: socket: Connection refused
giu 19 12:40:07 raspberrypi AstroPhoto-Plus[445]: socket: Connection refused
giu 19 12:40:09 raspberrypi AstroPhoto-Plus[445]: Latest version: AstroPhotoPlus-1.0.1-b923-Ubuntu_Debian.deb
giu 19 12:40:09 raspberrypi AstroPhoto-Plus[445]: Current version: 1.0.1-b923
giu 19 12:40:11 raspberrypi AstroPhoto-Plus[445]: tigervncserver: No vncserver running for this user!
giu 19 12:40:11 raspberrypi AstroPhoto-Plus[445]: tigervncserver: No vncserver running for this user!
giu 19 12:40:28 raspberrypi AstroPhoto-Plus[445]: Dispatch command error(-1): INDI: Could not find property CCD_TEMPERATURE in ZWO CCD ASI120MM Mini
giu 19 12:40:28 raspberrypi AstroPhoto-Plus[445]: <setNumberVector device="ZWO CCD ASI120MM Mini" name="CCD_TEMPERATURE" state="Idle" timeout="60" timestamp="2020-06-19T10:40:28">
giu 19 12:40:28 raspberrypi AstroPhoto-Plus[445]:     <oneNumber name="CCD_TEMPERATURE_VALUE">
giu 19 12:40:28 raspberrypi AstroPhoto-Plus[445]: 23.699999999999999289
giu 19 12:40:28 raspberrypi AstroPhoto-Plus[445]:     </oneNumber>
giu 19 12:40:28 raspberrypi AstroPhoto-Plus[445]: </setNumberVector>
giu 19 12:40:28 raspberrypi AstroPhoto-Plus[445]: Dispatch command error(-1): INDI: Could not find property CCD_EXPOSURE in Canon DSLR EOS 1300D
giu 19 12:40:28 raspberrypi AstroPhoto-Plus[445]: <setNumberVector device="Canon DSLR EOS 1300D" name="CCD_EXPOSURE" state="Idle" timeout="60" timestamp="2020-06-19T10:40:28">
giu 19 12:40:28 raspberrypi AstroPhoto-Plus[445]:     <oneNumber name="CCD_EXPOSURE_VALUE" min="0.00025" max="3600" step="1">
giu 19 12:40:28 raspberrypi AstroPhoto-Plus[445]: 1
giu 19 12:40:28 raspberrypi AstroPhoto-Plus[445]:     </oneNumber>
giu 19 12:40:28 raspberrypi AstroPhoto-Plus[445]: </setNumberVector>
giu 19 12:40:29 raspberrypi AstroPhoto-Plus[445]: Dispatch command error(-1): INDI: Could not find property CCD_TEMPERATURE in ZWO CCD ASI120MM Mini
giu 19 12:40:29 raspberrypi AstroPhoto-Plus[445]: <setNumberVector device="ZWO CCD ASI120MM Mini" name="CCD_TEMPERATURE" state="Idle" timeout="60" timestamp="2020-06-19T10:40:28">
giu 19 12:40:29 raspberrypi AstroPhoto-Plus[445]:     <oneNumber name="CCD_TEMPERATURE_VALUE">
giu 19 12:40:29 raspberrypi AstroPhoto-Plus[445]: 23.699999999999999289
giu 19 12:40:29 raspberrypi AstroPhoto-Plus[445]:     </oneNumber>
giu 19 12:40:29 raspberrypi AstroPhoto-Plus[445]: </setNumberVector>
giu 19 12:40:29 raspberrypi AstroPhoto-Plus[445]: Dispatch command error(-1): INDI: Could not find property CCD_EXPOSURE in Canon DSLR EOS 1300D
giu 19 12:40:29 raspberrypi AstroPhoto-Plus[445]: <setNumberVector device="Canon DSLR EOS 1300D" name="CCD_EXPOSURE" state="Idle" timeout="60" timestamp="2020-06-19T10:40:28">
giu 19 12:40:29 raspberrypi AstroPhoto-Plus[445]:     <oneNumber name="CCD_EXPOSURE_VALUE" min="0.00025" max="3600" step="1">
giu 19 12:40:29 raspberrypi AstroPhoto-Plus[445]: 1
giu 19 12:40:29 raspberrypi AstroPhoto-Plus[445]:     </oneNumber>
giu 19 12:40:29 raspberrypi AstroPhoto-Plus[445]: </setNumberVector>
giu 19 12:40:30 raspberrypi AstroPhoto-Plus[445]: [2020-06-19 12:40:30 +0200] [565] [WARNING] Error on property added: Property initialization error: Canon DSLR EOS 1300D/status/, label=None, type: None

GuLinux, 2020-06-19:
> giu 18 20:37:45 raspberrypi AstroPhoto-Plus[445]: setting log level from WARNING to WARNING
Sorry, still not enough information.
As you can see here, log level is still in WARNING mode. I need it to be on debug mode, you can find the setting to change it on the Settings page

ettoreribaudo, 2020-06-19:
raspberrypi:~ $ sudo AstroPhotoPlus-ctl logs
-- Logs begin at Thu 2019-02-14 11:11:58 CET. --
giu 19 13:19:01 raspberrypi systemd[1]: Starting Astrophotography sequence server...
-- Subject: L'unità AstroPhotoPlus.service inizia la fase di avvio
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- L'unità AstroPhotoPlus.service ha iniziato la fase di avvio.
giu 19 13:19:01 raspberrypi systemd[1]: Started Astrophotography sequence server.
-- Subject: L'unità AstroPhotoPlus.service termina la fase di avvio
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- L'unità AstroPhotoPlus.service ha terminato la fase di avvio.
-- 
-- La fase di avvio è done.
giu 19 13:19:01 raspberrypi AstroPhoto-Plus[427]: + GUNICORN_BINARY=gunicorn3
giu 19 13:19:01 raspberrypi AstroPhoto-Plus[427]: + which gunicorn3
giu 19 13:19:01 raspberrypi AstroPhoto-Plus[427]: + GUNICORN_BINARY=gunicorn
giu 19 13:19:01 raspberrypi AstroPhoto-Plus[427]: + gunicorn --threads 4 -t 30 -k gthread -b 0.0.0.0:5000 api_entrypoint:app --log-level warning --log-syslog --log-syslog-prefix AstroPhotoPlus --disable-redirect-access-to-syslog
giu 19 13:19:03 raspberrypi AstroPhoto-Plus[427]: **** Initialising Redis server on port 45609
giu 19 13:19:10 raspberrypi AstroPhoto-Plus[427]: setting log level from WARNING to WARNING
giu 19 14:30:04 raspberrypi AstroPhoto-Plus[427]: socket: Connection refused
giu 19 14:30:04 raspberrypi AstroPhoto-Plus[427]: socket: Connection refused
giu 19 14:30:06 raspberrypi AstroPhoto-Plus[427]: Latest version: AstroPhotoPlus-1.0.1-b923-Ubuntu_Debian.deb
giu 19 14:30:06 raspberrypi AstroPhoto-Plus[427]: Current version: 1.0.1-b923
giu 19 14:30:08 raspberrypi AstroPhoto-Plus[427]: tigervncserver: No vncserver running for this user!
giu 19 14:30:08 raspberrypi AstroPhoto-Plus[427]: tigervncserver: No vncserver running for this user!
giu 19 14:30:30 raspberrypi AstroPhoto-Plus[427]: setting log level from WARNING to DEBUG
giu 19 14:30:30 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:30:30 +0200] [559] [INFO] setting [log_level] updated: WARNING => DEBUG
giu 19 14:30:33 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:30:33 +0200] [559] [DEBUG] check_active_connections: active_connections=1
giu 19 14:30:33 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:30:33 +0200] [559] [DEBUG] check_active_connections: connected or accesspoint activated
giu 19 14:30:37 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:30:37 +0200] [559] [INFO] BLOBClient: INDI server connected
giu 19 14:30:37 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:30:37 +0200] [559] [DEBUG] BLOBClient connected
giu 19 14:30:37 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:30:37 +0200] [559] [INFO] update property: Astrometry/CONFIG_PROCESS ({'CONFIG_LOAD': True})
giu 19 14:30:37 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:30:37 +0200] [559] [INFO] update property: Astrometry/CONNECTION ({'CONNECT': True})
giu 19 14:30:37 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:30:37 +0200] [559] [INFO] setting property value for CONFIG_PROCESS: {'CONFIG_LOAD': True}
giu 19 14:30:37 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:30:37 +0200] [559] [DEBUG] on_switches: ['CONFIG_LOAD']
giu 19 14:30:37 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:30:37 +0200] [559] [DEBUG] off_switches: []
giu 19 14:30:37 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:30:37 +0200] [559] [INFO] setting property value for CONNECTION: {'CONNECT': True}
giu 19 14:30:37 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:30:37 +0200] [559] [DEBUG] on_switches: ['CONNECT']
giu 19 14:30:37 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:30:37 +0200] [559] [DEBUG] off_switches: []
giu 19 14:30:37 raspberrypi AstroPhoto-Plus[427]: 2020-06-19T12:30:37: No cameras detected.Check power and make sure camera is not mounted by other programs and try again.
giu 19 14:30:37 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:30:37 +0200] [559] [INFO] update property: ZWO CCD ASI120MM Mini/CONFIG_PROCESS ({'CONFIG_LOAD': True})
giu 19 14:30:37 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:30:37 +0200] [559] [INFO] setting property value for CONFIG_PROCESS: {'CONFIG_LOAD': True}
giu 19 14:30:37 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:30:37 +0200] [559] [DEBUG] on_switches: ['CONFIG_LOAD']
giu 19 14:30:37 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:30:37 +0200] [559] [DEBUG] off_switches: []
giu 19 14:30:37 raspberrypi AstroPhoto-Plus[427]: 2020-06-19T12:30:37: No cameras detected.Check power and make sure camera is not mounted by other programs and try again.
giu 19 14:30:37 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:30:37 +0200] [559] [INFO] update property: ZWO CCD ASI120MM Mini/CONNECTION ({'CONNECT': True})
giu 19 14:30:37 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:30:37 +0200] [559] [INFO] setting property value for CONNECTION: {'CONNECT': True}
giu 19 14:30:37 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:30:37 +0200] [559] [DEBUG] on_switches: ['CONNECT']
giu 19 14:30:37 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:30:37 +0200] [559] [DEBUG] off_switches: []
giu 19 14:30:37 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:30:37 +0200] [559] [INFO] update property: PMC8/CONFIG_PROCESS ({'CONFIG_LOAD': True})
giu 19 14:30:37 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:30:37 +0200] [559] [INFO] setting property value for CONFIG_PROCESS: {'CONFIG_LOAD': True}
[... cut ...]
giu 19 14:32:05 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:32:05 +0200] [559] [DEBUG] check_active_connections: active_connections=1
giu 19 14:32:05 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:32:05 +0200] [559] [DEBUG] check_active_connections: connected or accesspoint activated
giu 19 14:32:35 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:32:35 +0200] [559] [DEBUG] check_active_connections: active_connections=1
giu 19 14:32:35 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:32:35 +0200] [559] [DEBUG] check_active_connections: connected or accesspoint activated

ettoreribaudo, 2020-06-19:
pi@raspberrypi:~ $ sudo AstroPhotoPlus-ctl logs
-- Logs begin at Thu 2019-02-14 11:11:58 CET. --
giu 19 14:35:10 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:35:10 +0200] [559] [DEBUG] check_active_connections: active_connections=1
giu 19 14:35:10 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:35:10 +0200] [559] [DEBUG] check_active_connections: connected or accesspoint activated
giu 19 14:35:35 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:35:35 +0200] [559] [INFO] update property: PMC8/DEBUG ({'DISABLE': True, 'ENABLE': False})
giu 19 14:35:35 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:35:35 +0200] [559] [INFO] setting property value for DEBUG: {'DISABLE': True, 'ENABLE': False}
giu 19 14:35:35 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:35:35 +0200] [559] [DEBUG] on_switches: ['DISABLE']
giu 19 14:35:35 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:35:35 +0200] [559] [DEBUG] off_switches: ['ENABLE']
giu 19 14:35:37 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:35:37 +0200] [559] [INFO] update property: PMC8/USEJOYSTICK ({'ENABLE': True, 'DISABLE': False})
giu 19 14:35:37 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:35:37 +0200] [559] [INFO] setting property value for USEJOYSTICK: {'ENABLE': True, 'DISABLE': False}
giu 19 14:35:37 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:35:37 +0200] [559] [DEBUG] on_switches: ['ENABLE']
giu 19 14:35:37 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:35:37 +0200] [559] [DEBUG] off_switches: ['DISABLE']
giu 19 14:35:42 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:35:42 +0200] [559] [DEBUG] check_active_connections: active_connections=1
giu 19 14:35:42 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:35:42 +0200] [559] [DEBUG] check_active_connections: connected or accesspoint activated
giu 19 14:35:44 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:35:44 +0200] [559] [INFO] update property: PMC8/USEJOYSTICK ({'DISABLE': True, 'ENABLE': False})
giu 19 14:35:44 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:35:44 +0200] [559] [INFO] setting property value for USEJOYSTICK: {'DISABLE': True, 'ENABLE': False}
giu 19 14:35:44 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:35:44 +0200] [559] [DEBUG] on_switches: ['DISABLE']
giu 19 14:35:44 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:35:44 +0200] [559] [DEBUG] off_switches: ['ENABLE']
giu 19 14:35:50 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:35:50 +0200] [559] [INFO] update property: PMC8/TELESCOPE_TRACK_STATE ({'TRACK_ON': True, 'TRACK_OFF': False})
giu 19 14:35:50 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:35:50 +0200] [559] [INFO] setting property value for TELESCOPE_TRACK_STATE: {'TRACK_ON': True, 'TRACK_OFF': False}
giu 19 14:35:50 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:35:50 +0200] [559] [DEBUG] on_switches: ['TRACK_ON']
giu 19 14:35:50 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:35:50 +0200] [559] [DEBUG] off_switches: ['TRACK_OFF']
giu 19 14:35:51 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:35:51 +0200] [559] [INFO] update property: PMC8/TELESCOPE_PARK ({'UNPARK': True, 'PARK': False})
giu 19 14:35:51 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:35:51 +0200] [559] [INFO] setting property value for TELESCOPE_PARK: {'UNPARK': True, 'PARK': False}
giu 19 14:35:51 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:35:51 +0200] [559] [DEBUG] on_switches: ['UNPARK']
giu 19 14:35:51 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:35:51 +0200] [559] [DEBUG] off_switches: ['PARK']
giu 19 14:36:14 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:36:14 +0200] [559] [DEBUG] check_active_connections: active_connections=1
giu 19 14:36:14 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:36:14 +0200] [559] [DEBUG] check_active_connections: connected or accesspoint activated
giu 19 14:36:24 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:36:24 +0200] [559] [DEBUG] Looking up for 7000 on catalog NGC with Redis key catalog_entry_NGC-7000
giu 19 14:36:46 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:36:46 +0200] [559] [DEBUG] check_active_connections: active_connections=1
giu 19 14:36:46 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:36:46 +0200] [559] [DEBUG] check_active_connections: connected or accesspoint activated
giu 19 14:37:18 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:37:18 +0200] [559] [DEBUG] check_active_connections: active_connections=1
giu 19 14:37:18 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:37:18 +0200] [559] [DEBUG] check_active_connections: connected or accesspoint activated
giu 19 14:37:49 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:37:49 +0200] [559] [DEBUG] check_active_connections: active_connections=1
giu 19 14:37:49 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:37:49 +0200] [559] [DEBUG] check_active_connections: connected or accesspoint activated
giu 19 14:38:20 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:38:20 +0200] [559] [DEBUG] check_active_connections: active_connections=1
giu 19 14:38:20 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:38:20 +0200] [559] [DEBUG] check_active_connections: connected or accesspoint activated
giu 19 14:38:24 raspberrypi AstroPhoto-Plus[427]: INDI server localhost/7624 disconnected.
giu 19 14:38:24 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:38:24 +0200] [559] [INFO] indi server disconnected; disconnect_requested: 1592570304.4721403
giu 19 14:38:24 raspberrypi AstroPhoto-Plus[427]: INDI server localhost/7624 disconnected.
[... cut ...]
giu 19 14:39:12 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:39:12 +0200] [559] [DEBUG] on_switches: ['CONFIG_SAVE']
giu 19 14:39:12 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:39:12 +0200] [559] [DEBUG] off_switches: ['CONFIG_DEFAULT', 'CONFIG_LOAD', 'CONFIG_PURGE']
giu 19 14:39:20 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:39:20 +0200] [559] [DEBUG] check_active_connections: active_connections=1
giu 19 14:39:20 raspberrypi AstroPhoto-Plus[427]: [2020-06-19 14:39:20 +0200] [559] [DEBUG] check_active_connections: connected or accesspoint activated

GuLinux, 2020-06-20:
Hi,
I'm not seeing anything related to platesolving in your logs, and in your screenshots I'm not seeing whether you've actually started the platesolving process.
Maybe there's a misunderstanding about how this works.. it's not just goto, it's platesolving assisted goto.
You need to set a target, and start shooting with a camera. The telescope will do an initial move, then as soon as you get an image solved, the telescope sync and will keep adjusting until you're precisely on target.

ettoreribaudo, 2020-06-20:
OK i try tonight if clear.