GuLinux/AstroPhoto-Plus #253: lighttpd  instead of nginx?

aruangra, 2021-02-11:
Is it possible to run Astrophoto plus with lighttpd instead of nginx? 
Thanks.

gstrauss, 2021-02-11:
After looking at `config/dev-webserver-conf/nginx/astrophotoplus.conf`, a lighttpd equivalent might be as simple as this (untested):
```
server.port = 3080
server.name = localhost
server.max-request-size = 204800
server.max-write-idle = 65535
server.stream-response-body = 2
server.modules += ("mod_setenv", "mod_proxy")
setenv.set-response-header = (
  "Cache-Control" => "no-store, no-cache, must-revalidate, proxy-revalidate, max-age=0"
)
proxy.server = (
  ""     => ( ( "host" => "127.0.0.1", "port" => "3000" ) )
  "/api" => ( ( "host" => "127.0.0.1", "port" => "5000" ) )
)
```
However, `proxy.header` might need to be added if URL-mapping is needed between the request and the backend.

GuLinux, 2021-03-27:
Hi,
closing as this is not really relevant to AstroPhoto Plus.
Any HTTP server would work, providing it's properly configured, including lighttpd.
The configuration provided from @gstrauss looks correct to me.
Thanks

closed 2021-03-27 by GuLinux